phrack/ShootOFF-legacy #118: Calibration shapes do not move when the projection area is resized

cbdmaul, 2015-03-21:
To reproduce: Full screen projection area.  Enable calibration.  Calibration shapes will appear.  Un-full-screen the projection area.  The square will be missing.

Background: I have been switching between full screen and windowed attempting to diagnose other issues, but I have to end calibration and restart it every time I adjust the window size.

phrack, 2015-03-22:
The solution here depends on what we do about #121.

phrack, 2015-03-26:
I'll fix this one in 2.1 given that #121 was a wontfix.

phrack, 2015-03-26:
I am giving up on this one and unassigning the milestone. I'll leave it open though. I spent more than an hour on this. I tried half a dozen things that make complete logical sense given my 12 years of programming experience and extensive reading of this controls documentation. Half of them ended up doing nothing useful and the other half did extremely strange and unpredictable crap. Googling just turns up people whining about even more Tkinter canvas bugs.