phrack/ShootOFF-legacy #121: Projection does not seem to pick up calibration square even when clearly visible

cbdmaul, 2015-03-21:
I have attached a really terrible example.  This may be related to the overscan issue, although I wouldn't think that losing a small percentage of the square would cause it to not identify it.

Putting the window in windowed mode does cause the square to be identified part of the time, but it is still not anywhere near as identified as well as the triangle.

The right side of my projection area does have more background ambient light, I will test it again in the dark.

[image]

phrack, 2015-03-22:
This calibration process is annoyingly fragile and it doesn't achieve its intended goals as a result. The typical way to fix annoyingly fragile processes is to simplify. What if I just ripped out the entire calibration process and replaced it with a virtual target that gets added/remove when you open/close the arena. You align the virtual target with the projection just like you would any virtual target with its real target. Shots on the projector target would get translated to a shot on the projector. This would be a WAY more simple and and reliable I think. Worth it?

cbdmaul, 2015-03-22:
The problem with that is it would be pretty much impossible for me with my current setup to adjust a target to line up with the projection.  I obviously don't have an ideal setup but I'm already lying on the floor to use the laptop I'm running it on.

Also, the only way I can tell if the lighting and projection settings are appropriate is to check the calibration.  I can't tell if the camera is picking up the whole display without it.  I can't tell if the projection is too bright unless I have white objects to display, although I guess a target would work for that too.

phrack, 2015-03-22:
OK, so the primary hangups seem to be:

1. Aligning a virtual target on a projection would be a PITA without an external camera.
2. Without the shapes it's hard to see bounds and external settings.

1 isn't really ShootOFF's problem in any case :-p. For 2, what about instead when you hit calibrate the same shapes show up and you get a virtual target to align? When you stop calibrating the virtual target for calibration disappears (just gets hidden) and the shapes disappear.

cbdmaul, 2015-03-22:
I guess there's an overlooked issue here.  Is it worth being able to manually cause calibration?  If the lighting conditions are not satisfactory for the calibration mechanism as is, is the shot detection algorithm even going to work reliably?  I'd rather have the calibration fail and tell me, "fix your setup" then just make it a manual process and then I have no way of knowing when it is fixed.

It's much easier to adjust projection settings until the calibration detects the shapes then it would be to guess and check.

phrack, 2015-03-22:
The best way to figure that out is to project, point the camera at the projection, and see if you get a bunch of noise. Don't calibrate. It's possible calibration is way more sensitive than shot detection. If it is, there is no reason to do it automatically because all the calibration noise wouldn't tell you anything.

cbdmaul, 2015-03-22:
Seems like you'd be punishing all the people it works fine for just because of my crappy setup.

phrack, 2015-03-22:
It only would be if calibration tells you something useful about noise. My point is that if pointing the camera at the projection without calibrating doesn't detect a bunch of non-shots, you aren't gaining anything by having a sensitive calibration process. Basically see this branch in our decision tree right now:

1. If a bunch of non-shots are detected by just pointing the camera at the projection, this is a wontfix. 
2. If pointing the camera at the projection does not detect false positive shots, but does get true positives, the calibration process should be manual.

cbdmaul, 2015-03-26:
It does have trouble picking up the square, it'll flicker between finding it and not.  Even when shot detection is perfect in regards to detection accuracy.  But manually aligning a target would be a huge pain in my ass and the calibration still works even if it isn't perfect.

phrack, 2015-03-26:
I've had this happen without keystone issues too. My current theory is that it has to do with refresh rates either at the projector or the camera, and the rectangle washing out periodically as a result in a way that the human eye doesn't percieve. I say that because when it's happened to me I lose and regain the lock with a consistent frequency.

Anyway, based on what you said, I am marking this as won't fix here. I think we will have to revisit this in NG.

closed 2015-03-26 by phrack