phrack/ShootOFF-legacy #126: Stability issues - Crash

cbdmaul, 2015-03-23:
Duel target training causes crash.  Was able to reproduce by clicking it when there is no duel target in the projector area.  No python exceptions

[image]

phrack, 2015-03-23:
As in #125, I suspect it's a problem with Python. Maybe see if there is a newer version or try 64-bit? These actions aren't causing these issues on Ubuntu or Arch.

phrack, 2015-03-23:
On Arch: 

$ python2 --version
Python 2.7.9

cbdmaul, 2015-03-23:
it is 2.7.9.  to try 64bit i have to reinstall all the python packages.

phrack, 2015-03-23:
I know that feel. Is the crash new to 2.1 after some of the recent commits, or is this the first time you've tried these features on this system? Trying to establish if this is because of something new in ShootOFF or if it's something unhelpable in the environment.

cbdmaul, 2015-03-23:
It is a new environment for testing.  Previously I had been testing on a Win7-64 environment.  This is a Win8.1-64.  I did notice that audio is not working.  My guess is that one of the packages, maybe pyttsx, is either not installed properly or does not like something about how it is being initialized.

I'm going to switch all the packages to 64bit and try it again.

cbdmaul, 2015-03-24:
Audio works and it's not crashing with all 64bit libraries

I haven't tested everything but you can close these two issues and I'll open a new one for anything new.

closed 2015-03-24 by phrack