phrack/ShootOFF-legacy #127: Should be able to choose to show hits on projector area

cbdmaul, 2015-03-24:
For one, it would make it much easier to adjust the shot detection sensitivity when setting up a projector.  I could tell immediately if it doesn't pick up a shot.

But also, I would like to know where I'm hitting without having to look at the pc.  Same goes for training/duel scores.

I realize that making dots the same color as the laser may lead to recursive false positives but using a color that can be ignored would work.

phrack, 2015-03-24:
Please open this one in NG:

https://github.com/phrack/ShootOFF-NG

I agree with you, but this would be enough new work I don't want to do it in a codebase that is being retired. Closed here.

phrack, 2015-03-24:
PS. It would probably be best to keep anything that is new work (i.e. not a bug fix) in NG for the same reason.

closed 2015-03-24 by phrack