phrack/ShootOFF-legacy #129: Hits on targets not registering

cbdmaul, 2015-03-26:
[image]

With calibrated projector arena, hits clearly on targets are not registering.  None of the shots shown registered a hit on the tree or the reset button.

I had trouble seemingly off and on and it didn't seem to be a projection alignment issue.  I could hit all around and on targets sometimes without getting a registered hit.

phrack, 2015-03-26:
This was reported to me via email as well. Out of curiosity, can you try click-to-shoot without the projector calibrated and see if it happens? Maybe it looks like there is a lock calibration wise but it resizes to 0 right before you hit "Stop Calibrating"?

cbdmaul, 2015-03-26:
That is very much possible.  I will try click-to-shoot.

phrack, 2015-03-26:
I am calling this fixed, but please go ahead and comment if something is still obviously out of whack and potentially fixable. I did three things:

1. If a lock is lost during calibration, ShootOFF will remember the last lock even when you hit "Stop Calibrate". The last lock is replaced with a new lock if a new lock is in fact achieved. This should solve the problem if our conjecture was right.

2. When you stop calibrating, if you never had a lock, ShootOFF will now warn you that you aren't going to have any shots on the projector arena detected.

3. When you stop calibrating, if you had a lock, ShootOFF will tell you want bounding box was detected for the arena on the command line. This would make debugging easier if this bug is still actually not fixed (I can re-open it easy if so).

commit 35a33989 (2015-03-26): Projector calibration is now more resiliant to failure to achieve a lock. This should fix #129
closed 2015-03-26 by phrack