phrack/ShootOFF-legacy #130: My animated targets don't animate :(

cbdmaul, 2015-03-26:
[image]
[image]

Well, my plates turned into...some kind of clam shelly oyster shape

And my pepper popper didn't move until i shot it 5 times.  then it almost fell.

phrack, 2015-03-26:
To be clear, you're not seeing a progression? It's from the first frame directly to the last one?

cbdmaul, 2015-03-26:
Pepper popper jumped from first to last, no progression

phrack, 2015-03-26:
This probably due to #113.

cbdmaul, 2015-03-26:
Is it really? These shots are being detected, and I'm not shooting quickly enough that animations should be skipped.

phrack, 2015-03-26:
I'll reserve judgment until I see the profiling numbers. All I know right now is this GAAH ARHHA FAAJA tkinter canvas is so buggy I don't think anyone could even pay me to use it again.

phrack, 2015-03-31:
Did they animate in your most recent testing?

cbdmaul, 2015-03-31:
Yes, in linux.  Although I did not test with high cpu usage.

But on Windows the plate rack images don't look right, am I correct? If you look at the screenshot above, that's what I was seeing on Windows.  The plates don't look like that with the correct animation on linux.

phrack, 2015-03-31:
The plate racks in the image above are showing intermediate frames in the animation, like the canvas isn't being refreshed internally.