phrack/ShootOFF-legacy #134: TCLError: image "pyimage18809" does not exist

cbdmaul, 2015-03-31:
I resized my duel tree and all hell broke loose.  It started registering hits on the white line at the bottom of the duel tree and it caused all kinds of problems.  Duel tree is now missing a target.

[image]

Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1489, in __call__
    return self.func(*args)
  File "shootoff.py", line 423, in reset_click
    self._projector_arena.reset()
  File "/home/ubuntu/ShootOFF/projector_arena.py", line 51, in reset
    self._canvas_manager.reset_animations()
  File "/home/ubuntu/ShootOFF/canvas_manager.py", line 399, in reset_animations
    self._canvas.itemconfig(region, image=self._image_regions_images[region][FIRST_PHOTOIMAGE_INDEX])
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 2371, in itemconfigure
    return self._configure(('itemconfigure', tagOrId), cnf, kw)
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1265, in _configure
    self.tk.call(_flatten((self._w, cmd)) + self._options(cnf))
TclError: image "pyimage18809" doesn't exist
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1489, in __call__
    return self.func(*args)
  File "shootoff.py", line 423, in reset_click
    self._projector_arena.reset()
  File "/home/ubuntu/ShootOFF/projector_arena.py", line 51, in reset
    self._canvas_manager.reset_animations()
  File "/home/ubuntu/ShootOFF/canvas_manager.py", line 399, in reset_animations
    self._canvas.itemconfig(region, image=self._image_regions_images[region][FIRST_PHOTOIMAGE_INDEX])
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 2371, in itemconfigure
    return self._configure(('itemconfigure', tagOrId), cnf, kw)
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1265, in _configure
    self.tk.call(_flatten((self._w, cmd)) + self._options(cnf))
TclError: image "pyimage18809" doesn't exist
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1489, in __call__
    return self.func(*args)
  File "shootoff.py", line 423, in reset_click
    self._projector_arena.reset()
  File "/home/ubuntu/ShootOFF/projector_arena.py", line 51, in reset
    self._canvas_manager.reset_animations()
  File "/home/ubuntu/ShootOFF/canvas_manager.py", line 399, in reset_animations
    self._canvas.itemconfig(region, image=self._image_regions_images[region][FIRST_PHOTOIMAGE_INDEX])
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 2371, in itemconfigure
    return self._configure(('itemconfigure', tagOrId), cnf, kw)
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1265, in _configure
    self.tk.call(_flatten((self._w, cmd)) + self._options(cnf))
TclError: image "pyimage18809" doesn't exist

cbdmaul, 2015-03-31:
It broke shot detection also

Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1489, in __call__
    return self.func(*args)
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 536, in callit
    func(*args)
  File "shootoff.py", line 158, in detect_shots
    self.handle_shot(laser_color, x, y)
  File "shootoff.py", line 187, in handle_shot
    (x - bbox[0])*x_scale, (y - bbox[1])*y_scale)
  File "/home/ubuntu/ShootOFF/projector_arena.py", line 23, in handle_shot
    if "_shape:image" in self._arena_canvas.gettags(region) and self._canvas_manager.is_transparent_pixel(region, x, y):
  File "/home/ubuntu/ShootOFF/canvas_manager.py", line 380, in is_transparent_pixel
    hit_location_color = self._image_regions_images[region][LAST_IMAGE_INDEX].getpixel((x, y))
  File "/usr/lib/python2.7/dist-packages/PIL/Image.py", line 980, in getpixel
    return self.im.getpixel(xy)
IndexError: image index out of range

phrack, 2015-04-04:
I am having a hell of a time duplicating this. I've even put on multiple plate racks with multiple dueling trees, resized and move all of them in different ways, turned on a training protocol, and it still works as expected w/ no exceptions.

cbdmaul, 2015-04-09:
Wasn't able to reproduce in newest code

closed 2015-04-09 by phrack