phrack/ShootOFF-legacy #136: "You missed 1 shoot targets" / "You missed 2 shoot targets"

cbdmaul, 2015-03-31:
Added "SimpleBullseye_small_five".  Enabled " Shoot/Don't shoot" training.

It just says those two phrases repeatedly and occasionally shows some red circles for 100ms or so.  I'm not sure what I'm supposed to be doing but in any case it would be impossible to do anything with it talking and making red circles constantly.

phrack, 2015-03-31:
It's supposed to add its own targets (red circles and dark red circles). You're supposed to shoot the red ones and not the darker ones within 10 seconds, then you get a new set of random targets. This still sounds like a fixable bug.

cbdmaul, 2015-03-31:
I don't think I could tell which was red and which was dark red, but in any case the protocol is definitely not working correctly.

phrack, 2015-04-04:
I am at a loss on this one too. Just can't dupe it no matter what combination of things I try. I even added every single target multiple times to both the webcam feed and the projector arena, then turned on the protocol, and it still works right.

cbdmaul, 2015-04-09:
Was able to reproduce.  Start training protocol.  Repeatedly click reset.  After a couple seconds it will start going nuts.

cbdmaul, 2015-04-09:
I have other issues with this protocol, don't seem worthy of their own issue tickets:

1. If there are different reds, I can't tell.  All the circles look identical.
2. The protocol says up to 3 targets but I have only ever seen 0, 1, or 2.
3. Targets are frequently at the very edge of the screen, making them useless.
4. Targets frequently heavily overlap.

I also got this exception during shoot-don't shoot:

Exception in thread say_thread:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/ubuntu/ShootOFF/training_protocols/protocol_operations.py", line 160, in _say
    self._tts_engine.iterate()
  File "/usr/local/lib/python2.7/dist-packages/pyttsx/engine.py", line 218, in iterate
    self.proxy.iterate()
  File "/usr/local/lib/python2.7/dist-packages/pyttsx/driver.py", line 222, in iterate
    self._iterator.next()
ValueError: generator already executing

phrack, 2015-04-09:
It's hard to say what is going on with the reds because the projector might make them look identical or the don't shoot ones may be invisible when projected. The graphics are gif files in training_protocols/shoot_dont_shoot. Maybe try opening one in paint or something to scribble on it and see what happens when you rerun the protocol? 

I am not concerned with overlapping, but I am conflicted about them being on the edge of the screen. I knew that happens because I got it enough with click to shoot, but to me it was basically just like something being behind cover. I wasn't sure how annoying or not that would be using an actual trainer. Is it the case that you regularly get ones you just can't shoot because they are mostly off the screen?

cbdmaul, 2015-04-09:
A good percentage are far enough to the edge to be at least annoying.

I just shut the projector down but I'll try editing the target colors at some point

phrack, 2015-04-09:
OK, I'll make it so that they won't go as far off screen so that even if they are "behind cover" there is still something to shoot. Will have to wait for you about colors since I can only make WAGs without a projector.

phrack, 2015-04-09:
This is my current tally on this one:

1. Sometimes freaks out in odd situations -- Unless there is more to this that I don't know, it's not really worth the time to fix a bug in retiring code base that happens when you do something weird like hit the reset button a bunch of times. Particularly given that this is largely due to bugs in a third-party library I already really annoyed with at this point (pyttsx).
2. If there are different reds, I can't tell. All the circles look identical. -- Waiting for feedback on this since it's likely due to don't shoot targets getting washed out on the projector.
3. The protocol says up to 3 targets but I have only ever seen 0, 1, or 2. -- FIXED: Will now generate up to 3 targets.
4. Targets are frequently at the very edge of the screen, making them useless. -- FIXED: They will no longer have more than 2/3rd's of the target off screen.
5. Targets frequently heavily overlap. -- This will depend on how well the PRNG is seeded. I am honestly not worried about it even if there is something that can be done, but I am also not convinced there is a great way to fix it given this PRNG (if it's really pseudo-random you can see clustering sometimes no matter what, even in ways that seem deterministic). 

TL; DR: 2 fixed, 2 wontfixes, and an up in the air.

cbdmaul, 2015-04-12:
The dark red is just not visible on this projector.  Shows up black, so I can't see those.

I wanted to edit the gif for it to confirm, but I downloaded 3 different editors on ubuntu and none of them have supported saving gifs.  In fact, just opening that gif caused problems in the ubuntu image viewer and Gnome Paint Drawing Editor.

Can we stop using gifs since I hate them? webm for the animated ones?

phrack, 2015-04-12:
We can try other formats in NG, but in the python version only gifs work for animations without having the animations be a lot more manual. I use GIMP. In GIMP if you have to click File -> Export to get a GIF. 

My personal opinion is that NG is coming along fast enough we should just fix this last color issue and then consider 2.1 ready for release. The rest of the known issues are things I am going to keep open because they are serious enough for people to know they are known or they are small things that are generally non-deterministic and minor when they manifest.

cbdmaul, 2015-04-13:
Yeah, the target colors are just identical on my projection setup.  Both shoot and don't shoot do show up.

phrack, 2015-04-13:
Can you please pick a different color for don't shoot that works on the projector? Go ahead and even push a new version of the GIF if you're comfortable with that, then we can this up and do the release.

bklaw, 2015-04-14:
I'll redo the colors. What colors are you needing?

phrack, 2015-04-14:
I don't know what they should be, but cbdmaul may have some insight. The problem is that commodity projectors wash out the color, so you'll have two very different shades of, say, red on your laptop that look identical on a projector. They can't look identical because you need to be able to tell the difference between the two semi-quickly.

cbdmaul, 2015-04-14:
Think it would be more logical to have "shoot" be like a green or grey and "dont shoot" be the red.  They can still remain different shades, should still allow color-blind use.

I had problems with the paint bucket in GIMP, I am probably an idiot

phrack, 2015-04-16:
I just changed the colors. Will package and release 2.1 later today.

commit 0e3c47a4 (2015-04-16): Shoot is now grey an don't shoot is now red. This closes #136
closed 2015-04-16 by phrack