phrack/ShootOFF-legacy #140: No score with ISSF protocol?

cbdmaul, 2015-04-09:
[image]

Don't seem to get any score for hits, maybe i'm doing it wrong.

phrack, 2015-04-09:
It should show the score on every hit. My guess is that it's not interacting with the arena correctly but would work if the target were instead added to the camera fees.

phrack, 2015-04-09:
God dammit, I can produce this one either. These last few bugs are *really* annoying :(. 

[image]

cbdmaul, 2015-04-12:
Not able to reproduce but I've got a different ISSF protocol bug.  Can shoot in a round more than 5 times.

[image]

cbdmaul, 2015-04-12:
My thought on the original bug is that I broke something with the don't shoot/shoot protocol that caused the zero scores in ISSF.

phrack, 2015-04-12:
This is a race condition, but one that I don't consider to be worth fixing in a retiring code base given that it isn't counting hits. I am going to call this issue closed.

closed 2015-04-12 by phrack