phrack/ShootOFF-legacy #141: AttributeError: 'super' object has no attribute 'init' (OS X 10.10)

s0nspark, 2015-04-27:
I worked through the install instructions on my MacBook running OS X 10.10 and get the referenced error when running... I have all of the requirements installed:

    $ sudo pip install -r requirements.txt
    Requirement already satisfied (use --upgrade to upgrade): numpy>=1.7.1 in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from -r requirements.txt (line 1))
    Requirement already satisfied (use --upgrade to upgrade): pillow>=2.2.1 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 3))
    Requirement already satisfied (use --upgrade to upgrade): pyttsx>=1.1 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 4))
    Requirement already satisfied (use --upgrade to upgrade): pyttk>=0.3.2 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 5))
    Requirement already satisfied (use --upgrade to upgrade): pyaudio>=0.2.7 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 6))

Here is the stack trace:

    $ python ./shootoff.py
    Traceback (most recent call last):
      File "./shootoff.py", line 902, in <module>
        mainWindow = MainWindow(config)
      File "./shootoff.py", line 855, in __init__
        self._protocol_operations = ProtocolOperations(self._webcam_canvas, self)
      File "/Users/Tim/Downloads/phrack-ShootOFF-0e3c47a/training_protocols/protocol_operations.py", line 34, in __init__
        self._tts_engine = pyttsx.init()
      File "/Library/Python/2.7/site-packages/pyttsx/__init__.py", line 39, in init
        eng = Engine(driverName, debug)
      File "/Library/Python/2.7/site-packages/pyttsx/engine.py", line 45, in __init__
        self.proxy = driver.DriverProxy(weakref.proxy(self), driverName, debug)
      File "/Library/Python/2.7/site-packages/pyttsx/driver.py", line 66, in __init__
        self._driver = self._module.buildDriver(weakref.proxy(self))
      File "/Library/Python/2.7/site-packages/pyttsx/drivers/nsss.py", line 24, in buildDriver
        return NSSpeechDriver.alloc().initWithProxy(proxy)
      File "/Library/Python/2.7/site-packages/pyttsx/drivers/nsss.py", line 28, in initWithProxy
        self = super(NSSpeechDriver, self).init()
    AttributeError: 'super' object has no attribute 'init'

Any help appreciated...

phrack, 2015-04-27:
This is a but in pyttsx. A ShootOFF user fixed it but didn't contribute the fix upstream. Here is what they said:

"I made a change to /usr/local/lib/python2.7/site-packages/pyttsx/driver/nsss.py

from Foundation import *
from AppKit import NSSpeechSynthesizer
from PyObjCTools import AppHelper
from ..voice import Voice
# Added by JNR 2/23/2015 to fix deprecated super problem
from objc import super
#

The builtin super does not work with objc, thus the change."

Alternatively, you could wait for NG, which should have far fewer cross-platform issues: 

https://github.com/phrack/ShootOFF-NG

s0nspark, 2015-04-27:
Thanks - that did the trick... and thanks for mentioning NG! I wasn't aware
of the new project :)

phrack, 2015-04-27:
No problem! If you're a dev, feel free to join us! Plenty of open issues to tackle.

closed 2015-04-27 by phrack